space-concordia-robotics/robotics-prototype #19: Users shall be able to see whether a motor is alive or dead, or has errors

PeterGhimself, 2018-03-20:


vashmata, 2019-07-28:
The only way we would be able to guess if a motor is working or not would be through some logical leaps using the current measurements from this issue: https://github.com/space-concordia-robotics/robotics-prototype/issues/17 but it's not ideal.

PeterGhimself, 2019-10-10:
Hey @vashmata what do you think of this description in the use case? @MartensCedric and I are auditing the issues, I wanna know if you think this is a reasonable approach to getting the statuses of motors in general

vashmata, 2019-10-12:
I think it's generally ok but we should clarify dead/alive with error. If the motor isn't turning and the current draw is lower than expected then we know it's dead or there's a broken wire. If the motor isn't turning and the current draw is way too high then it's dead or there's a shorted wire. Etc. And cases where the encoder is working or not, cases where the current is in hte expected range but the motor speed isn't matching i

PeterGhimself, 2019-10-13:
@vashmata good points. I feel like there _may_ even be some unforeseen results that we will only find out during testing

closed 2022-10-31 by VimFreeman