space-concordia-robotics/robotics-prototype #22: Users shall be able to view the camera feed

PeterGhimself, 2018-03-20:


bnduwi, 2018-05-22:
@ZaynMuhammad I think you are working on this issue. When your ready you can use Zenhub (chrome extension) to move it to in progress.

bnduwi, 2018-07-07:
Tested out the code on branch https://github.com/space-concordia-robotics/robotics-prototype/issues/13.
OpenCV setup on windows was went ok.
Ran the StreamCatcher.py program and saw the camera feed from my laptop.
Pressing q did close the feed.
Good job Zayn

bnduwi, 2018-07-07:
Since we'll eventually switch to using Flask I started looking at how it can be done. Found this link: https://blog.miguelgrinberg.com/post/flask-video-streaming-revisited
The code for it on this github: https://github.com/miguelgrinberg/flask-video-streaming

I tested it out and it was able to provide a camera feed in the browser.
[image]

[image]

Above is view of running it using cmd since an environment variable has to be set to use opencv
i.e: set CAMERA=opencv (in Windows) 

Next image shows viewing the feed in browser (note have to use 125.0.0.1 instead of feedback loop as mentioned in cmd image)

gabrielkarras, 2018-09-21:
For issue #22 and #23, you'll need to install OpenCV on your machine as we'll be using it to develop/work with the camera. Follow this installation tutorial (link below). We'll proceed to import Zayn's code for GTK to Flask (my task).
 
_Installation tutorial_: https://medium.com/@debugvn/installing-opencv-3-3-0-on-ubuntu-16-04-lts-7db376f93961.

PeterGhimself, 2018-11-20:
Work started on branch [view-camera-feed-22](/space-concordia-robotics/robotics-prototype/tree/view-camera-feed-22)

PR #70 (2018-12-01): View camera feed 22
closes #22 

A very simple implementation allowing us to view the arm camera stream, given that the `StreamDispatcher.py` process is running on the rover.
The methods/libraries may have to be changed since mjpeg streams aren't generally supported past the`img` tag and using the newer HTML `video` tag would give us more to work with.
This implementation does _not_ implement starting the stream or stopping it via commands sent from the base station, nor updating FPs. The stream must be initiated manually from the rover itself.

Tested on a single server/client ubuntu 16.04 ma [... cut]
closed 2018-12-01 by PeterGhimself