space-concordia-robotics/robotics-prototype #28: Users shall be able to reset the motors to home position

PeterGhimself, 2018-03-20:


vashmata, 2019-03-31:
It looks like I did a lot of work on this under the tag for manual control (#24) because I forgot we actually had a separate issue for this.... So this comment is just to indicate that homing is being worked on (and has basic functionality) in branch 24

PeterGhimself, 2019-10-10:
@vashmata are you working on this in some branch somewhere? If so which one. Also in your checklist your last point "Command to move to neutral position" what do you mean by _neutral_ ? Is it the same as home position?

vashmata, 2019-10-12:
Currently this is not attached to any branch. I'm defining neutral position as 0 0 0 0 0 0 and home position as the resting configuration of the arm, some configuration that looks "natural" as a starting point for competition tasks.

PeterGhimself, 2019-10-13:
what are the advantages of having these two distinct positions to go to? Without proper encoder feedback I see how having a command to go to neutral helps us "reset" our open loop style estimated angle values, but is it as relevant with proper feedback from the motors?

vashmata, 2019-10-13:
There isn't really an advantage to bringing the arm to "neutral" once the angles are all correct. That said this raises the question of how we should home the arm properly:
- if the whole process is automated, how do we guarantee that the arm won't accidentally collide with the ground or the rover? We'd need vision or sensors or something for self-collision and environment-collision avoidance
- otherwise my 3 ideas are:
- - we home the joints one at a time and adjust the arm's configuration as we go to ensure that it doesn't collide with anything
- - we bring the whole arm into the neutral position and then set all the joint software angles to zero
- - similar to the previous point but we go joint by joint (i.e. set one joint to 0 then reset that joint's software angle, then move to the next joint)

So basically neutral doesn't really make much sense to go to once the arm angles are known. The home position may or may not be useful. Maybe I need to rewrite this stuff a little bit.

PeterGhimself, 2019-11-25:
[image] [A2.2 Homing and Inverse Kinematics](https://trello.com/c/gfcSHcZL/143-a22-homing-and-inverse-kinematics)

closed 2022-10-31 by VimFreeman