space-concordia-robotics/robotics-prototype #36: ROS model

PeterGhimself, 2018-04-03:


bnduwi, 2018-12-05:
[image]

Steps to reproduce:
- Follow this tutorial on generating a ROS package with URDF from SolidWorks:
-- To install solidworks addin (plugin): http://wiki.ros.org/sw_urdf_exporter
-- To create ROS package for a SolidWorks assembly: http://wiki.ros.org/sw_urdf_exporter/Tutorials/Export%20an%20Assembly (Have to define the joints to add the URDF file)
-- Export the assembly to generate the ROS package. (Name it in lower cases letters with underscores as spaces)
-- To run simulation in Gazebo copy the ROS package in ROS workspace `~/catkin_ws/src`
-- Next build the package by first `cd ~/catkin_ws` and the running `catkin_make` command
-- Running the simulation is supposed to be done in one command but there is an issue so follow these steps:
--- cd into catkin_ws folder and then run `roslaunch <package name> gazebo.launch` (This command will present some errors but the model of the arm has been generated and spawned.
--- To view the model simulation open a separate terminal window and run `roslaunch gazebo_ros empty_world.launch`

PeterGhimself, 2019-10-10:
Closing via #134

closed 2019-10-10 by PeterGhimself