space-concordia-robotics/robotics-prototype #39: GUI Styling

tatumalenko, 2018-05-22:
# GUI Styling
We should take a few moments to talk about styling of the GUI. It's obviously not pressing, but depending on our opinions, implementation could be as easy as two lines +- of code. 

Styling currently employed is very colourful. I propose going for a nice, clean, ui look that is akin to Mac users. This is an example of how the ui would look without much doing honestly to the default GTK styling built-in:
[image]

Obviously it's pretty darn vanilla looking, but nonetheless, with a few accentuated styling choices like adding more of that exact same blue to different actions (mouseover or button pressed, etc), I think it would make for a sweet as look.

I can see some wanting more colour, maybe a colour theme that relies on one principle colour and applying it in various tones. This idea is exemplified and highly encouraged by @PeterGhimself (lel) in the following image:
[image]

What does every think? Let the discussions begin.

bnduwi, 2018-05-25:
I think this is a good idea. Definitely the UI needs some revamping. It current looks as is because of experimentation to see how things work in GTK.

I like the single color with different tones idea and also the hover over color change would be great. Another thing but really not important is layout to maybe make look more like a dashboard with the controlls in the middle and a lot bigger for easier view. Just a thought ...

tatumalenko, 2018-05-26:
What do you guys think of [this](https://github.com/dpgaspar/Flask-AppBuilder) Python Flask app look?

PeterGhimself, 2018-07-29:
I think we should have our own predefined themes (set up default one).
Would be cool to also let the user create their own custom themes as well.

vashmata, 2018-09-01:
Closing via #42, #55

closed 2018-09-01 by vashmata