space-concordia-robotics/robotics-prototype #40: GUI running on Mac scaling issue caused by matplotlib

tatumalenko, 2018-05-22:
# GUI running on Mac scaling issue caused by matplotlib
There seems to be an issue for Mac users when trying to run the GUI with the matplotlib Figure Canvas component loaded. I found a similar user report of this on [Stack Overflow](https://stackoverflow.com/questions/43980607/embedding-matplotlib-messes-up-gtk3-dpi) (but with no replies sadface). I tested with and without the matplotlib component and here is the result.

With matplotlib component (full screen view, still cut off some part of it - no layout resizing when window resizing occurred):
[image]

Without matplotlib component (would scale appropriately to the window's size):
[image]

For now this isn't a big deal since we're mostly using Linux setups anyways, but just good to keep in mind for those wanting to work on a Mac setup instead.

PeterGhimself, 2018-07-22:
Closing as we are moving away from GTK (to flask)

closed 2018-07-22 by PeterGhimself

tatumalenko, 2018-08-24:
This might be irrelevant, but I might have found out why MacOS had issues with `matplotlib`. Unrelated, but might of been the fix ([link here](https://stackoverflow.com/questions/21784641/installation-issue-with-matplotlib-python)) necessary.