space-concordia-robotics/robotics-prototype #56: [Rover] GPS localization map

PeterGhimself, 2018-09-01:
Create a map in the rover GUI page that will be able to always show the rover's position.

vashmata, 2019-07-29:
The current GUI has a placeholder image with 3 dots representing the antenna, the rover and a goal. These could be used in a rudimentary localization map or a totally different approach could be taken.

vashmata, 2019-08-17:
[image] [Implement Interactive Map](https://trello.com/c/GQFc6EB7/86-implement-interactive-map)

MartensCedric, 2019-10-11:
[image] [GPS Localization Map](https://trello.com/c/4rwnEdNr/153-gps-localization-map)

MartensCedric, 2020-02-07:
After discussing with @vashmata and @mattsklivas we have established a MVP for URC2020 including the issues #288, #289 and #290. The GPS map would be a simple canvas with the rover at the center. It would provide information on the direction of the goals. This would be used in the Extreme Retrieval Task

AFknUsername, 2020-03-09:
does this need to be a "map" (like google maps) or can it be an abstract rendering of obstacles and stuff? To scale and with accurate locations of course

MartensCedric, 2020-03-09:
@AFknUsername What we discussed me, @vashmata and @mattsklivas. In the first iteration of this map (Basically everything that is covered by this epic and added to the URC milestone), we don't care about terrain data. It can literally be a white background with a triangle looking rover where goals/waypoints are just dots. This first map iteration is meant for the operators to get a general feel of where to go.

AFknUsername, 2020-03-09:
@MartensCedric ok I understand, I will implement it right away since I already have it working

closed 2022-10-31 by VimFreeman