space-concordia-robotics/robotics-prototype #58: Embed a ROS dummy arm mirror into Flask

bnduwi, 2018-09-17:
This issue is to get a dummy ROS simulation/mirror embedded into the current flask UI.

Research has been done and there were three options found:
1) Use ros3djs which results in [this](http://wiki.ros.org/ros3djs/Tutorials/VisualizingAURDF)
2) Use wviz ros library which results in [this](http://wiki.ros.org/wviz/Tutorials/Video%20Tutorial%20-%20Introduction%20to%20wviz)
3) Use Gazebo's web component reffered to as Gzweb which is seen [here](http://gazebosim.org/gzweb.html)

For this issue option one is selected and implemented.

bnduwi, 2018-09-22:
Currently using the the steps in this [link ](http://wiki.ros.org/ros3djs/Tutorials/VisualizingAURDF) and also modify the linked javascript to:

`<script src="https://static.robotwebtools.org/threejs/r88/three.min.js" async></script>
    <script src="https://static.robotwebtools.org/threejs/r88/ColladaLoader.js" async></script>
    <script src="https://static.robotwebtools.org/threejs/r88/STLLoader.js" async></script>
    <script src="https://static.robotwebtools.org/EventEmitter2/0.4.14/eventemitter2.min.js" async></script>
    <script src="https://static.robotwebtools.org/roslibjs/0.20.0/roslib.min.js" async></script>
    <script src="https://static.robotwebtools.org/ros3djs/current/ros3d.min.js" async></script>
    <script src="../static/js/ros_model.js" async></script>`

will be able to run the simulation in a separate web page. However the simulation does not get rendered correctly at all.

The other issue is that if it is actually embedded in our GUI it does not show up at all. Probably some conflicts with all the javascript or css files our GUI currently has. Needs to troubleshooting.

vashmata, 2019-08-25:
[image] [Arm simulation in GUI](https://trello.com/c/NxVFuJlv/126-arm-simulation-in-gui)