space-concordia-robotics/robotics-prototype #86: Science page

PeterGhimself, 2019-02-15:
Wireframe:

[image]

bnduwi, 2019-05-14:
First draft visual layout
[image]

PeterGhimself, 2019-08-14:
[image] [Test Drill Controls](https://trello.com/c/F2CF07CD/82-test-drill-controls)

vashmata, 2019-08-17:
[image] [Test turntable through GUI](https://trello.com/c/Yo3vsijE/83-test-turntable-through-gui)

vashmata, 2019-08-17:
[image] [Link Turntable to GUI](https://trello.com/c/1618zHa5/6-link-turntable-to-gui)

PR #163 (2019-09-23): Science page 86
Ready for URC/integration testing.
I've done my best to test these changes both locally through the GUI with a teensy and the drill/elevator/turntable rig (AKA the barbiehouse) and made sure to test adding the video stream feature again with the spare odroid we have.
I'm going to make further templatifying the science page a different issue as there are more pressing matters to be worked on for ERC still and the science page is functional as it is.

Here's a screenshot of me testing the camera stream on the science page.
[image]

Closes #86
closed 2019-09-23 by PeterGhimself