space-concordia-robotics/robotics-prototype #99: Integrate ROS directly inside of GUI

vashmata, 2019-05-15:
## TODO checklist:
- [ ] Modify webpage to use ROS directly instead of using workarounds
- - [x] Make a demo webpage which interfaces with ROS
- - [ ] Refactor app.py to behave in the same way
- - [ ] Add guidelines for starting up the ros websockets server along with app.py
- [ ] Build a ROS package for interfacing with Arm Teensy
- - [ ] Add rosbridge_websocket to dependencies wherever that needs to be
- - [ ] Build ROS packages for interfacing with other teensies or make the ROS package work for all of them
- - [ ] Create or find a .srv file which sends a String and responds with a String (the response) and a bool (indicating success or failure)
- - [ ] Make serial class to easily interface with serial port either via usb or uart
- - - [ ] Implement timeout for listening to serial port for desired feedback on opening serial comms

PeterGhimself, 2019-05-15:
You can check out the [.srv](https://github.com/space-concordia-robotics/robotics-prototype/blob/master/robot/rospackages/src/ping_acknowledgment/srv/PingResponse.srv) file from the ping_acknowledgment package I wrote. It's just missing the boolean in the response.

PeterGhimself, 2019-07-05:
Closing via #102

closed 2019-07-05 by PeterGhimself