space-concordia-robotics/robotics-prototype #108: Joy stick motor control

PeterGhimself, 2019-06-10:
Demo to connect controlling any type of motor with arduino over ROS network using a joy stick controller such as XBOX 360/PS3.
Good place to start [here](https://github.com/space-concordia-robotics/robotics-prototype/tree/master/robot/rospackages/src/robot_teleop_joy).

Waleed-HM, 2019-07-10:
I changed the name to Joystick Controls only because I will also be doing arm and camera controls using the joystick

PeterGhimself, 2019-07-10:
Renaming back so as to keep this task more granular, another issue for joy stick arm controls will be made.

PeterGhimself, 2019-07-10:
Created and pushed branch [joystick-wheel-control-108](/space-concordia-robotics/robotics-prototype/tree/joystick-wheel-control-108) for this issue.

PR #166 (2019-09-06): Joystick wheel control 108
Closes #108
closed 2019-09-06 by vashmata