space-concordia-robotics/robotics-prototype #111: Local mode for rover roslaunch file

PeterGhimself, 2019-07-06:
Currently on startup the Odroid runs `rosRoverStart.sh` which calls `roslaunch rover.launch` amongst other things. This calls multiple server side scripts which would crash if run on the basestation since they depend on obscurities that only exist on the Odroid.
To make it easier for people to work locally a local roslaunch file should be created. The current services that don't have local modes should be modified to have them (not as command line arguments but as ros parameters).

- [ ] Change current local modes in server side scripts to use ros params instead of command line args (or both if no conflicts)
- [ ] Add local modes to all other services that don't yet have it but would need it
- [ ] Create the local rover roslaunch file
- [ ] Test/verify local rover roslaunch file

vashmata, 2019-08-17:
[image] [Fully implement local mode software setup](https://trello.com/c/it68DGPN/121-fully-implement-local-mode-software-setup)

vashmata, 2019-08-17:
[image] [Local ROS Mode](https://trello.com/c/97eWSoDM/52-local-ros-mode)

MartensCedric, 2019-10-11:
[image] [Local Mode](https://trello.com/c/smULSKMJ/159-local-mode)

closed 2022-10-31 by VimFreeman