space-concordia-robotics/robotics-prototype #112: Rover Status LED

PeterGhimself, 2019-07-13:
Using Odroid GPIO pins, have certain LED patterns to indicate readiness/different states.

MVP:
A process to monitor that rover ROS nodes are running.

Extra:
- Check on base station connection/ROS nodes
- Check connection to MCUs (which node is running, if said MCU is "activated" --> use "who" serial cmd to query)

PeterGhimself, 2019-10-10:
We need to redefine the reqs here based off the new rules for ERC _and_ URC 2020

PeterGhimself, 2019-11-25:
[image] [A14.5 LED Control Module](https://trello.com/c/rdBV7ueV/157-a145-led-control-module)

PeterGhimself, 2019-11-25:
@KoaTheKoala just to clarify. This task (for URC 2020) should cause the LED only to flash when we've reached the target at the end of a leg during autonomous traversal, correct? Because previously I had interpreted the competition rules to mean that when the autonomy task would start and we would run certain nodes _then_ to indicate that we are driving autonomously with the use of this LED.

MartensCedric, 2020-07-11:
@PeterGhimself Is the description of this issue still accurate

closed 2022-10-31 by VimFreeman