space-concordia-robotics/robotics-prototype #114: LLC unit test

PeterGhimself, 2019-07-20:
Code to facilitate testing for MUX LLC failures.

Script for Odroid:
1. Send TX data of "{currentChannel},OK"
2. Repeat above for all channels
3. Log everything to file

Sketch for Arduino:
1. Listen to RX (check if serial1,2,3,4 available)
2. If RX data available, print data to USB of "{currentChannel},OK", include timestamp if possible
3. Send TX "{currentChannel,OK}"

vashmata, 2019-08-17:
[image] [Automate MUX Test](https://trello.com/c/VufBVxr4/20-automate-mux-test)

PeterGhimself, 2019-10-10:
This is less relevant now that we are going to move away from UART, however for now the PDS is still using UART to communicate with the odroid and all that remains to be done is for this code to be properly tested. Having a functional test for the LLC/mux board wouldn't hurt.

PeterGhimself, 2019-10-10:
For convenience we should make sure that we have a spare:
- [ ] Odroid
- [ ] eMMC/SD card with good image
- [ ] At least 2 teensies

vashmata, 2019-10-12:
> This is less relevant now that we are going to move away from UART, however for now the PDS is still using UART to communicate with the odroid

Given that we won't be switching mux channels anymore I believe it is better to leave this and focus on new tasks.

PeterGhimself, 2019-10-13:
@vashmata I agree but since for legacy reasons we have 4 options, we can see them as redundancies... could be helpful to switch if something breaks and we have no time for a proper fix and have to hack it on the spot?
Although to be fair this test was meant to be more of an endurance test if I remember correctly... it was @abting 's idea after all :grinning:

closed 2022-10-31 by VimFreeman