space-concordia-robotics/robotics-prototype #117: PDS integration

PeterGhimself, 2019-07-26:
Linking the current PDS to the GUI through ROS, adding controls/feedback.
The PDS is able to report: battery voltage, rover wheel currents (6), and temperatures (3)

- [ ] Create ROS node to listen to PDS and publish to topic for GUI to use
- [ ] Integration test of the above (use ros bash cmds to verify)
- [ ] Add PDS feedback visualisation to GUI
- [ ] Add control cmds to GUI
- [ ] Final integration test

vashmata, 2019-07-28:
The issue https://github.com/space-concordia-robotics/robotics-prototype/issues/17 relates to this issue. The motor table which is supposed to display the motor currents exists inside the rover webpage.

vashmata, 2019-08-17:
[image] [PDS -> GUI Implementation](https://trello.com/c/YdBibq1o/54-pds-gui-implementation)

PR #164 (2019-09-14): Pds integration 117
Created PDS webpage, PdsNode with Nick's help, linked everything together and tested it.

Closes #117
closed 2019-09-14 by vashmata