space-concordia-robotics/robotics-prototype #118: USB hub integration

PeterGhimself, 2019-07-26:
Given that the current design assumed using a MUX with 4 UART devices, modify the GUI/rover code in such a way to work with the USB hub.

- [ ] Modify the GUI/rover task handler code to ensure that listeners work in USB mode
- [ ] Account for camera USB port naming scheme

vashmata, 2019-08-17:
[image] [USB Hub Proof of Concept](https://trello.com/c/LwGpEMwF/19-usb-hub-proof-of-concept)

PeterGhimself, 2019-08-24:
Omitting templatifying science page in this issue since it is unrelated and I will deal with that issue in branch [science-page-86](/space-concordia-robotics/robotics-prototype/tree/science-page-86)
Omitting handling multiple simultaneous video streams mainly for the same reason as for templatifying science page. New issue opened #151 for that purpose. :eyes:

PR #157 (2019-09-03): Usb hub integration 118
Closes #118 

Added new backup of odroid image [here](https://drive.google.com/drive/u/1/folders/12PI-sH2UVobUI4k0Et0AxUC4BJWMFUDr)
closed 2019-09-03 by PeterGhimself