space-concordia-robotics/robotics-prototype #121: Base station launch desktop icon

PeterGhimself, 2019-07-26:
Currently to launch the base station, we need to run `./app.py`, as well as `roslaunch rosbridge_server rosbridge_websocket.launch` (making sure to deactivate venv for the second command).

Pro-tip: After the first part works, you can use `ps aux | grep <keyword>` to find the process id, and kill it with `kill -9 <process_id>`

- [ ] Create icon to run above commands in one shot
- - [ ] Ensuring a way of killing processes gracefully on exit
- [ ] (Super nice to have) Piping the output of the above commands either to a log file or some other console

vashmata, 2019-07-28:
Side note, we have officially merged a sample `.bash_aliases` file inside `/basestation/config` which holds two aliases for the above commands. `startgui` runs the flask server (app.py) and `rosgui` runs the ROS websocket server (runs the launch file)

vashmata, 2019-08-17:
[image] [Desktop shortcut to launch the GUI](https://trello.com/c/W3faT0kJ/122-desktop-shortcut-to-launch-the-gui)

vashmata, 2019-08-17:
[image] [Base station Launch Icon](https://trello.com/c/hV7HuuiC/98-base-station-launch-icon)

MartensCedric, 2019-10-10:
todo for me (Cedric) : Unmerge the ROS3d branch

F

MartensCedric, 2019-11-11:
Putting this on hold until there's a resolution with the rosbridge/venv issue

closed 2022-10-31 by VimFreeman