space-concordia-robotics/robotics-prototype #122: IMU functionality

PeterGhimself, 2019-07-26:
We have on IMU that gives us a lot of information but we're only using the magnetometer.
All we can get from it is the heading.

- [ ] Make use of all the sensors (accelerometer + gyroscope)
- - [ ] Get roll, pitch and yaw
- - [ ] Look into sensor fusion (and/or Kalman Filter)
- [ ] Figure out calibration method given that IMU must be inside rover during calibration 
- [ ] Determine how to get better heading measurements (because currently it's inconsistent, offset by some value which is different in either direction, and we don't know why)

vashmata, 2019-07-28:
See this issue/branch for some previous work: https://github.com/space-concordia-robotics/robotics-prototype/issues/53

vashmata, 2019-08-17:
[image] [Relative Rover Positioning](https://trello.com/c/WbkeUiWS/11-relative-rover-positioning)

vashmata, 2019-08-17:
[image] [IMU Proof Of Concept](https://trello.com/c/30HztiLv/10-imu-proof-of-concept)

closed 2022-10-31 by VimFreeman