space-concordia-robotics/robotics-prototype #126: PDS code improvements

vashmata, 2019-07-28:
The current PDS microcontroller code does not make full use of the components on its PCB. Moving forwards, some goals should be determined and worked on in partnership with the elec team.

- [ ] Read PDS documentation and discuss code improvements with relevant elec team members
- [ ] Add code improvements to this checklist and work on them

PeterGhimself, 2019-10-10:
@vashmata what's up? Shall we brainstorm together for this on the upcoming workday? Please extend on your checklist

NichHarris, 2019-10-10:
@peterghimself @vashmata I believe there's a section about code improvements in the PDS doc if you want to take a look there for some ideas. I believe most of the suggestions have been completed, but there's still a few that haven't been looked at.

vashmata, 2019-10-12:
Yeah for future reference check in with Nick and Phil/Abtin about this issue

PeterGhimself, 2019-10-13:
@NichHarris could you link the doc here please?

NichHarris, 2019-10-13:
> @NichHarris could you link the doc here please?

Here it is: https://docs.google.com/document/d/1beHVFolN7XjGXdMtjlnHVk-39P7YqkVzFxjqlRinOEQ/edit#heading=h.34am14b77cb2

closed 2022-10-31 by VimFreeman