space-concordia-robotics/robotics-prototype #128: ROS javascript library does not always accept custom message types

vashmata, 2019-07-28:
While rosifying the GUI, certain message types were defined. For example, a lightweight message type for navigation data was created. However, roslibjs would not recognize the message type. Some internet searches revealed that this issue has been discovered by others as well.

In the meantime, default message types such as Point are being used instead. They pose no problems, it's just that having custom messages would make it more clear what certain messages are for.

PeterGhimself, 2019-10-10:
@vashmata still relevant? :eyes:

vashmata, 2019-10-12:
Still relevant.

PeterGhimself, 2019-10-13:
Since this issue seems to be known to the general public, is this issue a work around? Would this mean that a proper fix would mean that the roslibjs devs fix it on their end? Do we have the most up to date version of roslibjs/is the problem already resolved since this issue was opened?

Might be worth while to check [this](https://github.com/RobotWebTools/roslibjs) out :eyes:

AFknUsername, 2020-05-19:
I know this is old, but I believe my current PR #366 should solve any issues with ROS message types.
Maybe revive this and make sure that there are no more issues?

closed 2022-10-31 by VimFreeman