space-concordia-robotics/robotics-prototype #132: Link inverse kinematics to arm MCU

vashmata, 2019-07-29:
Inverse kinematics calculations already function and a node exists, but the GUI has no way to request the calculations or send the results to the arm MCU. The arm also needs to have the correct angle calculations.

- [x] The inverse kinematics calculator from branch/issue 91 was adjusted in branch 105
- [x] A ROS node was also created in branch 105
- [ ] Adjust variables in arm MCU code to get proper angle calculations
- [ ] (should potentially do this in a different branch) confirm that homing code works and use it for angle calculations
- [ ] Retest inverse kinematics ROS node and calculator to confirm they work
- [ ] Add inverse kinematics controls to GUI
- [ ] Link said controls to ROS
- [ ] Test the arm controls and adjust parameters

vashmata, 2019-08-17:
[image] [Inverse Kinematics](https://trello.com/c/zaDPo6r7/22-inverse-kinematics)

PeterGhimself, 2019-11-25:
[image] [A2.2 Homing and Inverse Kinematics](https://trello.com/c/gfcSHcZL/143-a22-homing-and-inverse-kinematics)

closed 2022-10-31 by VimFreeman