space-concordia-robotics/robotics-prototype #138: Redesign the navigation data template

MartensCedric, 2019-08-04:
Currently the navigation data template is a vomited html form. It should be redone in a way that it doesn't look like it was rushed for URC by @vashmata

steamypotato, 2019-10-12:
Things that will also be updated:
-Add functionality for multiple GPS destinations
-Add support for different long/lat formats (as needed by URC rules)

vashmata, 2019-10-12:
I recommend also looking into the CIRC rules to see if they have different rules/tasks related to navigation. I believe they have a similar thing with multiple GPS goals.

PeterGhimself, 2019-10-13:
So then [this](https://circ.cstag.ca/2020/tasks/#reactor-patrol-route-traversal-task) would be the relevant documentation to refer to for the CIRC traversal task

MartensCedric, 2019-11-11:
This is updated and now includes the addition of a new page. Perhaps we should call this autonomy and include everything with navigation and autonomy? Thoughts @PeterGhimself @KoaTheKoala @vashmata? Since there is a scope change we should redefine what this task actually entails, hopefully it doesn't lead to being an epic

PeterGhimself, 2019-11-11:
@MartensCedric how about if we confine this issue to making what we had look better on a new page.
We can create a new issue which will have to do with the design of the page in relation to autonomy (adding the UI for setting multiple points). I guess another issue could then also be for the map rendering/drawing component.

MartensCedric, 2019-11-11:
@steamypotato Could you define the scope of what you are planning to do. This is to avoid that your task can't be estimated and will last forever Lol. Anything else outside the scope, we would create a new issue.

Just from what I understand of that comment @PeterGhimself. This issue would be basically to have the navigation template look better and have better UX while being transferred to a new page? I think this is good as along as the features and components of that template are defined here

What are the features of this new template @steamypotato ?

Are they still these?

- [ ] Add functionality for multiple GPS destinations
- [ ] Add support for different long/lat formats (as needed by URC rules)

Or is there more to it?

PeterGhimself, 2019-11-11:
Yeah adding a few helpful things like the two you mentioned up there would be cool, we're definitely gonna need it either way.

KoaTheKoala, 2019-11-11:
I think we only need to be able to input one GPS coordinate at a time. After reaching the destination, we would then get the next one & send it to the rover

steamypotato, 2019-11-12:
@MartensCedric yes, doing those, but josh also pointed out the idea of selecting a coordinate by clicking on a locating from google maps (gonna do those two forsure, but I can't say forsure I'm gonna do the google map one, tbh that one might require a new issue on its own)

MartensCedric, 2019-11-12:
FYI the Google Maps API requires internet AFAIK. Definitely create a new
issue and put a placeholder for now

KoaTheKoala, 2019-11-12:
There may be a way to download offline maps

Sent via the Samsung Galaxy S8+, an AT&T 5G Evolution capable smartphone

MartensCedric, 2019-11-12:
Yes, I think so too. This would probably be a different API though. Anyway,
it should be a seperate issue.

PeterGhimself, 2019-12-07:
Alright so after some discussions with @steamypotato @KoaTheKoala @NinjaWeeaboo 
What will be done in this task:
- Improve UI of nav template because it looks like trash
- Add support for multiple GPS formats (ensure conversion to decimal degrees before passing to backend)
- Pop-up [bootstrap modal](https://getbootstrap.com/docs/4.0/components/modal/) element to support adding multiple GPS coordinates at once in a queue like fashion
- Create new page for navigation (call it navigation) and include nav template (multiple coordinates still supported here too)

steamypotato, 2020-02-04:
@PeterGhimself @MartensCedric  im just about done with this issue, i got everything on the new page working, but i couldn't really figure out how to also have it on the original rover page, its just way too much too cram into that tiny space, probably requires many fancy modals and whatnot, would you still say having those features on that rover page is still mandatory before making a PR for this issue?

MartensCedric, 2020-02-04:
Congratulations on this issue that you worked hard on! Now prepare to get roasted 😎

PR #287 (2020-09-26): Update nav 138
# Assignee Section

## Description

We have finished issue #138. READ the description to know what it is about. **It is NOT about creating a path system. It's just about making a basic GUI for navigation. Please do not request changes out of scope of this issue**. You can make new issues though.

closes #138

## Testing

0. `cd` into `robot/rospackages`, then `catkin_make` and then `sb`
1. `startgui`, then `rosgui`, and `roslaunch local_rover.launch` in "~/Programming/robotics-prototype/robot/util/rosRoverStart".
2. Open `localhost:5000/navigation`, set antenna parameters (longi [... cut]
closed 2020-09-26 by MartensCedric