space-concordia-robotics/robotics-prototype #143: Show latency/ping through an icon

MartensCedric, 2019-08-06:
Show the ping through an icon. Similar to what you see in most games. It would auto refresh every 4-5 seconds to display an accurate ping. This is an improvement over having to click each device to see if it pings or not through the chat

MartensCedric, 2019-12-31:
We have a function in the javascript to ping a microcontroller or to ping the Odroid (the computer on the rover).

Therefore, most of the stuff is already implemented. What you need to do in #143 is to create an html template with Flask/Jinja which will be loaded into the webpage. This template should be an icon that represents the strength of the signal. You will also need to add an AJAX request to update the icon depending on the latency.

MartensCedric, 2020-01-04:
After talking with @Waleed-HM we chose to give it only two connection pictures because the ping is not a good indicator of connection strength if we have a camera stream running for example. Thus we will keep it to "Has connection to the Odroid" and "Does not have connection to the Odroid"

MartensCedric, 2020-01-04:
Rewrite the `pingOdroid` function with this signature

`pingOdroid(timeout = SOME_DEFAULT_TIMEOUT, callback = () =>{})`

The behavior of the function is to call the callback with the arguments [(bool) Success, (number) Millis]

MartensCedric, 2020-01-04:
If the timeout is passed then it should call the callback with [false, timeout]
(There is an example of how to use a timer expiration to call a function in requestTask in the js folder of roslibjs somewhere

PeterGhimself, 2020-01-04:
I strongly suggest that we display the latency in ms every second or so, since we are already displaying this value in the console log when we use the existing pingOdroid button on the GUI, it shouldn't be too difficult to implement right?

vashmata, 2020-02-07:
I see that you removed the mcu ping stuff from this issue. Upon rereading this it reminded me that pinging the MCU could give us either the latency between the OBC and the MCU, OR the latency between the base station computer to the OBC to the MCU. And I didn't see any new issues related to pinging the MCU, so I'm just commenting this here for completeness' sake

closed 2022-10-31 by VimFreeman