space-concordia-robotics/robotics-prototype #145: Decouple RequestTask from GUI

MartensCedric, 2019-08-07:
The current method signature `function requestTask (reqTask, reqStatus, buttonID, callback, reqArgs = '')`

Can be reduced to a smaller signature
`function requestTask (reqTask, callback, reqArgs = '')`

The reqStatus and button ID removal will allow much cleaner code (See camera streams and other use cases)

PeterGhimself, 2019-08-14:
[image] [Develop simple autonomous traversal algorithm](https://trello.com/c/J6kOSido/99-develop-simple-autonomous-traversal-algorithm)

vashmata, 2019-08-14:
We discussed this in person but `reqStatus` is still required. Perhaps it could be combined with `reqArgs` however.

MartensCedric, 2019-10-12:
- [ ] Decouple GUI with communication layer
- [ ] Reduce number of arguments
- [ ] Document the function
- [ ] Remove magic numbers
- [ ] Keep functions small

closed 2020-02-16 by MartensCedric