space-concordia-robotics/robotics-prototype #150: Obtain the linear velocity of the rover

MartensCedric, 2019-08-21:
Using the PX4FLOW Smart Camera, obtain the linear velocity of the rover. 

According to this table : https://docs.px4.io/v1.9.0/en/sensor/px4flow.html#optics
and some extrapolation, I estimate the maximum measurable velocity to be around 0.59 m/s if the Smart Camera would be positioned at around 24.6 centimeters from the ground.

vashmata, 2019-08-21:
This ROS package may help but as you said it looks a bit old: https://github.com/cvg/px-ros-pkg

vashmata, 2019-08-21:
Measured the lens we have and its diameter looks like 8mm and not 16mm but the best way to know would be to move it around and see what the max speed we measure from it is

MartensCedric, 2019-08-24:
Further testing showed that this device is not meant to be used for a rover. The velocity that the device gives is way too off for our needs. Another device must be found

MartensCedric, 2019-10-10:
Closed because smart camera was shit

closed 2019-10-10 by MartensCedric