space-concordia-robotics/robotics-prototype #151: Multiple simultaneous camera streams

PeterGhimself, 2019-08-24:
Currently the way video streaming is implemented in the code base is that by design we restrict the odroid to only dispatching one stream at a time, so always the same link is used on the client to capture this stream. This issue would deal with modifying the backend as well as frontend to allow multiple streams to be on at the same time.
We know that with our ERC radios we get a bandwidth of ~90 Mbps (source: @Waleed-HM ) so it should be able to handle the amount of data being streamed over the rover network. If any issues arise due to this the streaming individual cameras at lower resolution could help mitigate those issues.

- [ ] Backend
- - [ ] Update task handler and start_ros_stream.sh` to be able to handle multiple streams at once
- [ ] Frontend
- - [ ] Update video stream panel so that each page (rover/science/arm) only requests it's own camera and not give options for the other ones
- - [ ] Add new view which will be solely for being able to display all the streams at the same time
- [ ] Optional
- - [ ] Include setting default resolution in `cv_camera` node instantiation
- - [ ] Add feature for request to start/restart node with specific resolution
- - [ ] Front end for start/restart node with specific resolution (dropdown list of reasonable options)

MartensCedric, 2019-09-23:
From my convo with @PeterGhimself 

> Another thing that I think is a bit messy is the placement of the camera rotation.
The problem is that the button is included in "ExtraControls" which is kind of messy considering we have a stream on every page and that the "Streams" page (3 streams" may require 3 of these buttons. So I propose that eventually these stream itself with the button to flip and take a screenshot is 1 template). This can be done in multiple streams issue. I would just comment if you agree

KoaTheKoala, 2019-09-23:
Just to clarify, you want the following to be bundled into the same html code:
The stream, the button to flip the stream, and a button to take a screenshot?

Koa

MartensCedric, 2019-09-23:
@KoaTheKoala , yup however it should be neatly placed. (right now those buttons take a lot of space. It should just be a rotate icon and a camera (screenshot) icon .

Cheers

KoaTheKoala, 2019-09-23:
Sure thing, I can do that.

Sent via the Samsung Galaxy S8+, an AT&T 5G Evolution capable smartphone

MartensCedric, 2019-10-13:
A few weeks ago we discussed this layout. Just adding it here @KoaTheKoala 
[image]

PeterGhimself, 2019-11-25:
[image] [A15.3 Multiple Streams Assessment](https://trello.com/c/28T69OPU/158-a153-multiple-streams-assessment)

PeterGhimself, 2019-12-27:
Alright so after doing some testing by modifying the resolution set in the `start_ros_stream.sh` script responsible for starting up video streams I think the main conclusions to make:
- 3 cameras work with our current setup but only at a measly 240p which looks too small in my opinion
- 2 cameras work with our current setup at 360p, which in my opinion is good enough
- Any resolution higher than that only works with one stream at a time, except 1080p which simply does not work at all

That being said, I propose the following approach moving forward:
I will test if I can simply keep the default at 360p and trade places of one port on the usb hub (so for example the one assigned to "Rear" camera) with one of the teensies usbs (like "Rover" teensy).
I think that's the smallest amount of change necessary then to in theory be able to see all streams at the same time. I'm not testing for multiple basestations viewing it, so I'm not setting out to determine that. Any thoughts @KoaTheKoala ?
I know when it will come time again for ERC we will have a requirement to get a minimum resolution on pictures and for dealing with that I would suggest that we make an issue to give the user an option to start a camera stream with any supported resolution option. Scope of #201 maybe?
Also, I noticed that when the blue light flickering happened the output from the task handler kept spamming: `VIDIOC_STREAMON: No space left on device` after which that same camera would not be able to be turned on again until after rebooting the system. We can either open a new issue to deal with this case or I can perhaps look into it and deal with it here in my PR #248 ,

### 3 cameras at 240p
[image]

### 2 cameras at 360p
[image]

### 1 camera at 480p
[image]

So far we've been doing 720p by default, pretty sure 1080p does not work at all.

PeterGhimself, 2019-12-29:
Front end account for local mode:
[image]

As well as competition mode (as you can see resizing the frontend to such a small scale causes overlap, decent work around could be solving #258 ):
[image]

closed 2019-12-30 by PeterGhimself