space-concordia-robotics/robotics-prototype #160: Redesign the servo code used in the frontend

MartensCedric, 2019-09-04:
This issue requires the assignee to design and implement a new way of handling the servos in `rover.js`.

This redesign is motivated by many violations of best software practices. These include : 

- Code of a "lower layer" coupled with code from the GUI layer
- Long functions with many arguments
- A lot of copy pasted code
- Use of Magic Numbers

The new design should be clean with proper design documentation.

MartensCedric, 2019-09-07:
@vashmata @PeterGhimself 

I present my design to revamp the servo code in the JS :

[image]

I tried to keep in mind the four problems outlined above in order to ensure my design is good.

**Magic Numbers**

By analyzing the code I came to the realization that most of the magic numbers came from input purposes. This is why my design proposes a new file called `input.js` with a bunch of input constants to represent new characters. It would also encapsulate the `keyState` array.

**GUI code coupled lower level code**
Adding to what me, @PeterGhimself and @KoaTheKoala discussed today concerning #141 (templatifiying the JS basically). The template `RoverMotorControls.html` would have its own javascript file called `RoverMotorControls.js` that is responsible of it. This means it would be this javascript file that is responsible to update the buttons with `showKeyPress()` and *not in the logic of the servo code*. This file is responsible for updating the servos and making sure the web page buttons are updated correctly.

**Long functions with many arguments**

This model removes the need to update the GUI through the servo code, also since the servos know their own `pwmVal` you don't need to pass it, neither do you need to pass all the keyboard inputs. Therefore the functions have a minimal amount of parameters sent.

**Array as a return value**

Instead of having this scheme of a `newCommand`, the positionServo knows if it requires an update with `bool requiresUpdate()` therefore the need of an array return value is not required. 

```
  let newCommand = false
  if (keyState[upKey]) {
    lightUp(upBtn)
    if (pwmVal > minPwm) {
      pwmVal -= positionServoIncrement
    }
    newCommand = true
  } else if (keyState[downKey]) {
    lightUp(downBtn)
    if (pwmVal < maxPwm) {
      pwmVal += positionServoIncrement
    }
    newCommand = true
  }
  return [newCommand, pwmVal]
```

Please express your opinions on the design.

MartensCedric, 2019-09-07:
I updated the design, please check this one : 
[image]

vashmata, 2020-01-11:
I can't see any obvious flaws with the design you're presenting. Doesn't mean there aren't any, but I personally would have to actually write out the code in order to see if any issues popped up. That being the case I'm fine with someone trying it out.

Only thing I have to say is that it would be good for developers to be capable of adjusting the servo default pwm values during testing/integration. I can't recall how I dealt with that in the ERC version of the code but if there's a way to set it from the console or in the GUI itself it can be useful.

closed 2022-10-31 by VimFreeman