space-concordia-robotics/robotics-prototype #180: Make a redundancy when `env.js` fails

MartensCedric, 2019-09-27:
Currently, the generated ip by `env.sh` may fail to connect with a VM (@mattskv). A redundancy should be present to ensure something always works. Alternatively, finding a way to accommodate the VMs could also solve this problem.

mattskv, 2019-10-06:
It turns out issues with the website were because of PING_THROTTLE_TIME = 1000 in app.js (line 6) which led to the error TypeError: invalid assignment to const `PING_THROTTLE_TIME' in console. Commenting out this line in app.js led to no issues when interacting with the GUI.

MartensCedric, 2019-10-06:
> It turns out issues with the website were because of PING_THROTTLE_TIME = 1000 in app.js (line 6) which led to the error TypeError: invalid assignment to const `PING_THROTTLE_TIME' in console. Commenting out this line in app.js led to no issues when interacting with the GUI.

This is unrelated to this issue. It was introduced by accident when the science page templatification was merged. And I fixed it in my PR #186 which has yet to be accepted

closed 2022-10-31 by VimFreeman