space-concordia-robotics/robotics-prototype #182: Proper Input handling

MartensCedric, 2019-09-27:
Currently the input handling is not well designed. In the issue #160, I suggested a centralized approach where we would include `input.js`. The completion of this issue requires to create a file called `input.js` which is able to keep track of the different input states. This file should also have many constants which would be useful. Alternatively, find a library that does it already. The code should entirely be refactored with these new changes

MartensCedric, 2019-09-29:
- [ ] add listeners
- [ ] clear listeners
- [ ] constants available for all keys we use
- [ ] warn on key used for many purposes (optional)

akhilrajkiz, 2019-10-12:
```
input.addSimultaneousListener('P', 'CTRL', 'ALT', function(){
  printSomething();
})
```

```
input.add=Listener('P', function(){
  ChangeGUItoRed();
})

```

akhilrajkiz, 2019-10-12:
//In input

```
function addListener(key, function)
{
num = convertKeyToInt(key)

arrayOfFunctions[num] = function
}

document.addEventListener('keydown', function(event){
num = event.code
executeFunction((arrayOfFunctions[num])
})
```

akhilrajkiz, 2019-10-12:
https://en.wikipedia.org/wiki/Observer_pattern

closed 2022-10-31 by VimFreeman