space-concordia-robotics/robotics-prototype #192: Implement SCP to transfer final image from rover to host computer

MartensCedric, 2019-10-12:


PeterGhimself, 2019-11-25:
This is in relation to the epic #113 
I think this should be blocked by the panorama feature actually working first.

MartensCedric, 2019-11-25:
It is true that it is in relation to epic #113 . It is not required to be blocked as the script is to simply show that you can transfer an image from rover to host. This task can be done in parallel with any other.

PeterGhimself, 2019-11-25:
On second thought, in theory it should not have to actually depend on that. But I have set #135 as a blocker before work starts here. We should at least have set in stone where panorama snapshots are saved so that mock pictures can be organized properly for testing and working on this issue.

MartensCedric, 2019-11-25:
@PeterGhimself this makes sense. But let's be honest that #135 is quite trivial. @badgerwild could you also do #135 too?

badgerwild, 2019-11-25:
@MartensCedric @PeterGhimself I'll take care of #135 .

badgerwild, 2020-09-11:
Following a discussion with @PeterGhimself , renamed issue to reflect a more general application of the script since it can be used for transferring any file type.

closed 2022-10-31 by VimFreeman