space-concordia-robotics/robotics-prototype #204: Use CMake to build the C++

MartensCedric, 2019-10-31:
CMake should be configured to build with the same compiler as the Arduino IDE.

MartensCedric, 2019-12-14:
I'm stuck with this PR.

I get this message when running cmake CMakeLists.txt in `robot`

CMake Error: CMake can not determine linker language for target: science-atmega32.elf
CMake Error: Cannot determine link language for target "science-atmega32.elf".

steamypotato, 2020-03-22:
Scope of the issue that's planned
- [ ] Create an ELF binary for each .ino file in the respective directory, in a /bin
- [ ] The above should be done with a single invocation of a 'cmake' command, from a root directory.

Using cmake for uploading will be discussed at a further time, until the above is done

closed 2020-06-05 by steamypotato