space-concordia-robotics/robotics-prototype #206: JS load error

PeterGhimself, 2019-11-02:
The problem is most likely some dependencies between different js files and the fact that even listeners only get bound properly to html elements when they are loaded in a certain order.

To reproduce:

- Checkout [this](https://github.com/space-concordia-robotics/robotics-prototype/tree/8a7f01de909f3555d7d5fd3a57895a98974f2a7c) version of the repo
- Run gui with `startgui` and go to rover page at `localhost:5000/rover`
- Click any of the extra controls buttons and they will not work

If you remove the first and last line (the `$(document).ready` part) it will work again. This is only a temporary hack/fix.

MartensCedric, 2019-11-03:
The issue is caused because `$(document).ready()` doesn't wait for images to load. Instead `$(window).load()` fixes the issue

MartensCedric, 2019-11-03:
My guess is that onReady executes the code but the TitaToggleButton isn't ready yet so the click listener doesnt attach properly unless it was called after (hence why onLoad works)

PR #210 (2019-11-04): Jsload 206
Closes #206
closed 2019-11-04 by MartensCedric