space-concordia-robotics/robotics-prototype #209: CLI rover control

PeterGhimself, 2019-11-03:
Right now we have two main ways of moving the rover: using keyboard with the GUI or joystick with the GUI. We should have a 3rd mean independant of the GUI which will work on keypress. Implementation for input handling can be based off of [here](https://github.com/space-concordia-robotics/robotics-prototype/blob/master/robot/demos/RemoteRoverControl/ClientSender.py)

PeterGhimself, 2019-12-29:
According to @KoaTheKoala there are 6 things from the autonomys team perspective that we want to have out of this:

- Go forward or backwards with a variable speed
- Turn left or right with variable speed
- Full stop
- Slow stop

To clarify, the stops have nothing to do with PDS i.e cutting power to motors.

vashmata, 2020-02-07:
> According to @KoaTheKoala there are 4 things from the autonomys team perspective that we want to have out of this:
> 
> * Go forward or backwards with a variable speed
> * Turn left or right with variable speed

What do you think about requesting that the rover drive x metres/degrees or x seconds rather than simply driving until a command is sent to stop driving, @KoaTheKoala ? Has this option been discussed?

MartensCedric, 2020-08-08:
I would strongly suggest that the interface between the commands and the RoverNode should use Twist command. This would make our rover work in Gazebo. The unified interface would make us avoid copying code for the joystick and the gui. We would have access to abundent ROS related tools by avoiding reinventing the wheel.

MartensCedric, 2020-08-10:
Please do not implement actual CLI controls when you can just use the package http://wiki.ros.org/teleop_twist_keyboard

MartensCedric, 2020-08-10:
I suggest we rescope the issue to use twist commands

MartensCedric, 2020-08-13:
> I suggest we rescope the issue to use twist commands

Actually, no. This issue should be the action server that interfaces between autonomy commands and the RoverNode

MartensCedric, 2021-01-09:
This is done and works because Twist works with the rover now

closed 2021-01-09 by MartensCedric