space-concordia-robotics/robotics-prototype #211: Find equivalent or simulate a way to log info to ROS from roslibjs

MartensCedric, 2019-11-09:
Find a way to log front end logs into ros

PeterGhimself, 2020-01-04:
I think if we simply create a publisher to the topic `/rosout` it should have the same effect, though that might mean we have to format the message properly ourselves.

vashmata, 2020-02-07:
[here's](http://robotwebtools.org/jsdoc/roslibjs/current/index.html) the web documentation for reference, although I don't see anything about ROS logging. [I just made an issue](https://github.com/RobotWebTools/roslibjs/issues/352) in the roslibjs repo 😎 about it. Perhaps it's an opportunity to contribute to open source?

vashmata, 2020-02-10:
I managed to publish a message to /rosout from within the GUI but there's a bunch of data in the [/rosgraph_msgs/Log](http://docs.ros.org/api/rosgraph_msgs/html/msg/Log.html) message type that I don't yet know how to generate. If we want to do it nicely we could make the logwarn, logerr, loginfo etc functions just like you can do in rospy.

Here's my code. As you can see based on the message definition, when I created the message I left out a lot of data. I imagine that will have an impact on how ROS actually understands the log message and I'm not sure how much is necessary for it to work correctly (possibly all of it) but I don't know how to generate all the data that's in the definition.

```
ros_logger = new ROSLIB.Topic({
    ros: ros,
    name: 'rosout',
    messageType: 'rosgraph_msgs/Log'
  })
log_msg = new ROSLIB.Message({
    level:2,
    msg:"hello from the gui"
})
ros_logger.publish(log_msg)
```

vashmata, 2020-02-10:
we could make it like
```
// Severity level constants
// DEBUG=1 // debug level
// INFO=2  // general level
// WARN=4  // warning level
// ERROR=8 // error level
// FATAL=16 // fatal/critical level

function loginfo(message) {
    ros_logger.publish(
        new ROSLIB.Message({
            level : 2, // 2 is for 
            msg : message
        })
    )
}
```
or something

closed 2020-02-15 by PeterGhimself