space-concordia-robotics/robotics-prototype #220: Can't build cv_camera on a new setup

MartensCedric, 2019-11-11:
Following the current readme will lead to an error while building cv_camera. The fix is to require new users to install some dependencies that contain the .cmake files.

MartensCedric, 2019-11-11:
@KoaTheKoala , I remember you knew what those commands were. Do you still remember?

MartensCedric, 2019-11-11:
Apparently, these commands are 
```
sudo apt-get install ros-kinetic-cv-camera
sudo apt-get install ros-kinetic-web-video-server
```

MartensCedric, 2019-11-11:
Can you check @AFknUsername ?

AFknUsername, 2019-11-11:
It built succesfully now

MartensCedric, 2019-12-06:
Ah yes it does build for you @AFknUsername, but we should put the fix in the ReadMe.md (which is what the issue is about). As long as it isn't in the readme, people will be missing packages upon building

AFknUsername, 2019-12-06:
So, I need to add a section saying to install the dependencies before running cv_camera as a temporary fix?

MartensCedric, 2019-12-06:
Well the task isn't assigned to anyone. The readme file would be need to be updated with this 

```
sudo apt-get install ros-kinetic-cv-camera
sudo apt-get install ros-kinetic-web-video-server
```
It also needs to contain an explanation on why we need to install this stuff which can only be written by someone familiar with the camera stream components. So there's no action required on your side basically, the reason why you were tagged 25 days ago was to make sure that it fixed your problem, now the next step is updating the readme.

AFknUsername, 2019-12-06:
Ok, I see. I thought you forgot to close it. Didn't realize there was more to be done, but that makes sense.

MartensCedric, 2019-12-06:
@AFknUsername Now it is closed (I added the fix) and tbh it was kind of "high priority" because it makes every new member have a disfunctioning setup lol

PR #232 (2019-12-06): [#220] Clarify README and fix cv_camera issues
Closes #220
closed 2019-12-06 by MartensCedric