space-concordia-robotics/robotics-prototype #223: Ensure that the GUI is capable and designed to handle multiple operators

MartensCedric, 2019-11-13:
The GUI was not initially properly designed with the intent of allowing multiple operators. This task is to make sure that the GUI supports multiple operators.

MartensCedric, 2019-12-30:
So this issue is extremely vague, so pardon me for creating it. Although we need to investigate the current state of the GUI and plan our operations with multiple operators and ensure that everything is supported. Perhaps we should reword this issue differently. @PeterGhimself

PeterGhimself, 2019-12-31:
Multiple operators _can_ use the GUI at the same time. We know that only one operator may run the flask app and the rest need to go to the "host" ip address to visit the site.
The issue is we haven't fully tested the capacity of it beyond the ERC 2019 competition missions.
As well, in our space drive we should start writing up formal docs on step-by-step operations for each mission for each comp so it is crystal clear how things are gonna go down.
This is exactly what I did for us in ERC 2019 on a piece of paper. Each time before a mission I would just go through a little "trial" run and write down the steps like:
```
1. X runs flask server while Y pings the OBC
2. X opens rover page and runs preliminary checks, Y ssh's into OBC and does routine checks
```
(of course it was more detailed than the quick example above)

vashmata, 2020-02-07:
In addition, since @MartensCedric is working on #278 I'd like to bring up that running the arm URDF should also only occur on one operator's machine

closed 2022-10-31 by VimFreeman