space-concordia-robotics/robotics-prototype #228: Investigate NIRScan nano GUI functionality in Linux

PeterGhimself, 2019-11-27:
The only available versions of the GUI are for [Windows/Mac](http://www.ti.com/tool/DLPNIRNANOEVM).

This issue is to look into the feasibility of running the same software in linux, with the ultimate goal of being able to get the csv data and corresponding graphs when running a scan with the NIRscan nano.

PeterGhimself, 2019-11-27:
Found a relevant [thread](http://e2e.ti.com/support/dlp/f/94/t/688895?tisearch=e2e-sitesearch&keymatch=NIRscan%2520Linux) on the forums, will see if I can run the example code succesfully on Linux from my laptop, then on the odroid.

PhilCarPi, 2019-12-21:
[image] [A31.4 Comms](https://trello.com/c/nkCklSDk/98-a314-comms)

closed 2022-10-31 by VimFreeman