space-concordia-robotics/robotics-prototype #229: Improve voltage and temperature display with icons

MartensCedric, 2019-12-02:
Currently the navbar's right side looks like shit.

Voltage icon with current voltage.

Temperature icon with three values.

MartensCedric, 2019-12-02:
Also when the voltage or temperature goes beyond a critical value, the color of the text should be changed to something that pops

adrientremblay, 2020-01-04:
below or equal 12.5 V -> modal popup warning to turn robot off and disconnect BMS
over or equal to 16.9 V -> modal popup warning overcharged turn off, go discharge and disconnect BMS

No temperature sensor but valid range is 0-85 degrees

vashmata, 2020-01-11:
> below or equal 12.5 V -> modal popup warning to turn robot off and disconnect BMS
> over or equal to 16.9 V -> modal popup warning overcharged turn off, go discharge and disconnect BMS
> 
> No temperature sensor but valid range is 0-85 degrees

Keep in mind there should be a way for developers to easily disable this because sometimes the sensors happen to be unplugged or the microcontroller is off or something.

PeterGhimself, 2020-01-12:
@vashmata I think that just having a feature toggle in the code is enough, but I agree it's a good idea to include in this PR for the reasons you brought up.

PR #272 (2020-02-16): Display icons 229
# Assignee Section

## Description
Revamp the battery voltage and temperature indicators on the navbar of the GUI.  Also adds modal popups, colour changes, and a sound when voltage or temp is out of acceptable range.  Also upgrades from fontawsome 4 to 5.

closes #229 

Approval from at least one software team lead is necessary before merging.

# Reviewer Section

Aside from local testing and the General Integration Test it is implied that static analysis should be included in the verification process.

- [x] Local Test Performed Successfully
- [ ] [General Integration Test]( [... cut]
closed 2020-02-16 by PeterGhimself