space-concordia-robotics/robotics-prototype #230: Science page URC 2020

PeterGhimself, 2019-12-02:
Wireframe:
[image]

Only 8 cuvettes though on the turntable/carousel:
[image]

PeterGhimself, 2019-12-02:
I would suggest that we change the route of the old science page from `science` to `science/erc2019` and make the new one `science/urc2020`

PhilCarPi, 2019-12-21:
[image] [A31.5 GUI](https://trello.com/c/IDrRJzbN/202-a315-gui)

MartensCedric, 2020-10-10:
Good wireframe. Although it is a bit incomplete. I'll ask someone else to continue this

MartensCedric, 2020-10-10:
- There should be no excavate button
- There's no fan speed control
- There is 3 temp sensors
- Currents?

PeterGhimself, 2020-10-18:
The previous one was done using mockflow.com

If someone wants to work on this let me know and I'll send you my credentials so you can access the previous version. Can speed up time significantly.

closed 2022-10-31 by VimFreeman