space-concordia-robotics/robotics-prototype #236: Closed loop wheel control jitters

PeterGhimself, 2019-12-17:
When driving the rover using the current closed wheel control implementation, after driving and then stopping the motors seem to still be drawing current since we can hear them making their typical sounds.
The rover won't be driving away anytime soon from this while being idle in closed loop mode after initial drive commands have been sent, but clearly it is a problem since when no commands are sent the motors should not be drawing any current/trying to move.

- [ ] Investigate cause of jitters
- [ ] Implement fix

PeterGhimself, 2019-12-17:
I should add that when driving the rover with closed loop control it seems less finer than open loop. What I mean specifically is that using the same "speed" in closed loop moves the wheels faster/more than in open loop which makes it harder to do very small precise movements.

vashmata, 2020-01-11:
The best way to resolve this issue may simply be to do a review of the rover mcu code and potentially rewrite it.

vashmata, 2020-03-11:
based on discussions with @fauziame , he will first handle #127 , then create a new issue to adjust PID gains from the GUI, and finally try adjusting the rate at which the PID and the GUI work to find a sweet spot. I'll adjust the issue description.

closed 2022-10-31 by VimFreeman