space-concordia-robotics/robotics-prototype #246: Fix inconsistent video speeds from recording streams between different camera models

NinjaWeeaboo, 2019-12-20:
Currently ffmpeg appears to record video streams at a higher frame rate for many cameras, including the ones on the rover. This causing issues with the newly added record feed feature. Where the outputted video recorded from the rover is sped up.

MartensCedric, 2019-12-20:
Is it really different camera models or different laptops? @PeterGhimself (?) I don't think this has been investigated.

MartensCedric, 2019-12-20:
Assuming the issue is caused by different camera models perhaps the issue is that one is 30FPS thus when processing it uses 2 seconds worth of video to output 1 second of 60 FPS

NinjaWeeaboo, 2019-12-20:
I believe that it is a fps related issue, but I do need to look into it more.

NinjaWeeaboo, 2019-12-20:
It seems to be something to do with ffmpeg incorrectly detecting the fps of the video feed. A hacky way of fixing it, would be to force ffmpeg to always record at the same fps rate as the camera; but I want to avoid doing that. I think some more investigation is needed anyways to be certain.

PeterGhimself, 2019-12-20:
Based off this output from my testing it makes sense that this is FPS related since with the higher quality camera I get:

```frame=   28 fps=0.0 q=-1.0 size=    2386kB time=00:00:01.08 bitrate=18101.3kbits/s frame=   43 fps= 43 q=-1.0 size=    3663kB time=00:00:01.68 bitrate=17862.7kbits/s frame=   58 fps= 38 q=-1.0 size=    4944kB time=00:00:02.28 bitrate=17762.9kbits/s frame=   73 fps= 36 q=-1.0 size=    6222kB time=00:00:02.88 bitrate=17697.6kbits/s frame=   88 fps= 35 q=-1.0 size=    7502kB time=00:00:03.48 bitrate=17658.8kbits/s frame=  103 fps= 34 q=-1.0 size=    8778kB time=00:00:04.08 bitrate=17624.0kbits/s frame=  118 fps= 34 q=-1.0 size=   10056kB time=00:00:04.68 bitrate=17603.0kbits/s frame=  134 fps= 33 q=-1.0 size=   11422kB time=00:00:05.32 bitrate=17588.8kbits/s frame=  149 fps= 33 q=-1.0 size=   12705kB time=00:00:05.92 bitrate=17581.4kbits/s frame=  164 fps= 32 q=-1.0 size=   13984kB time=00:00:06.52 bitrate=17570.1kbits/s frame=  179 fps= 32 q=-1.0 size=   15263kB time=00:00:07.12 bitrate=17561.5kbits/s frame=  194 fps= 32 q=-1.0 size=   16543kB time=00:00:07.72 bitrate=17554.4kbits/s frame=  209 fps= 32 q=-1.0 size=   17820kB time=00:00:08.32 bitrate=17545.9kbits/s frame=  224 fps= 32 q=-1.0 size=   19099kB time=00:00:08.92 bitrate=17540.5kbits/s frame=  240 fps= 32 q=-1.0 size=   20462kB time=00:00:09.56 bitrate=17533.9kbits/s frame=  255 fps= 31 q=-1.0 size=   21741kB time=00:00:10.16 bitrate=17529.7kbits/s frame=  271 fps= 31 q=-1.0 size=   23103kB time=00:00:10.80 bitrate=17524.4kbits/s frame=  286 fps= 31 q=-1.0 size=   24383kB time=00:00:11.40 bitrate=17521.2kbits/s frame=  293 fps= 31 q=-1.0 Lsize=   24982kB time=00:00:11.68 bitrate=17521.4kbits/s```

vs lower quality:

```frame=   23 fps=0.0 q=-1.0 size=    1832kB time=00:00:00.88 bitrate=17056.9kbits/s frame=   31 fps= 28 q=-1.0 size=    2479kB time=00:00:01.20 bitrate=16921.8kbits/s frame=   39 fps= 24 q=-1.0 size=    3130kB time=00:00:01.52 bitrate=16866.9kbits/s frame=   47 fps= 22 q=-1.0 size=    3788kB time=00:00:01.84 bitrate=16862.9kbits/s frame=   55 fps= 20 q=-1.0 size=    4428kB time=00:00:02.16 bitrate=16793.5kbits/s frame=   63 fps= 20 q=-1.0 size=    5098kB time=00:00:02.48 bitrate=16838.2kbits/s frame=   71 fps= 19 q=-1.0 size=    5768kB time=00:00:02.80 bitrate=16876.0kbits/s frame=   79 fps= 18 q=-1.0 size=    6441kB time=00:00:03.12 bitrate=16912.4kbits/s frame=   87 fps= 18 q=-1.0 size=    7114kB time=00:00:03.44 bitrate=16941.0kbits/s frame=   95 fps= 18 q=-1.0 size=    7788kB time=00:00:03.76 bitrate=16968.5kbits/s frame=  103 fps= 17 q=-1.0 size=    8461kB time=00:00:04.08 bitrate=16988.3kbits/s frame=  111 fps= 17 q=-1.0 size=    9136kB time=00:00:04.40 bitrate=17010.1kbits/s frame=  119 fps= 17 q=-1.0 size=    9812kB time=00:00:04.72 bitrate=17029.2kbits/s frame=  127 fps= 17 q=-1.0 size=   10488kB time=00:00:05.04 bitrate=17047.5kbits/s frame=  135 fps= 17 q=-1.0 size=   11164kB time=00:00:05.36 bitrate=17062.2kbits/s frame=  143 fps= 17 q=-1.0 size=   11813kB time=00:00:05.68 bitrate=17038.0kbits/s frame=  151 fps= 17 q=-1.0 size=   12472kB time=00:00:06.00 bitrate=17029.0kbits/s frame=  159 fps= 17 q=-1.0 size=   13137kB time=00:00:06.32 bitrate=17028.2kbits/s frame=  167 fps= 16 q=-1.0 size=   13803kB time=00:00:06.64 bitrate=17028.9kbits/s frame=  175 fps= 16 q=-1.0 size=   14469kB time=00:00:06.96 bitrate=17029.9kbits/s frame=  183 fps= 16 q=-1.0 size=   15135kB time=00:00:07.28 bitrate=17030.7kbits/s frame=  191 fps= 16 q=-1.0 size=   15801kB time=00:00:07.60 bitrate=17031.6kbits/s frame=  196 fps= 16 q=-1.0 Lsize=   16218kB time=00:00:07.80 bitrate=17032.7kbits/s```

Also this makes me wonder if we should maybe look into getting and displaying current FPS of the video stream :thinking:

PeterGhimself, 2020-01-04:
@NinjaWeeaboo I just tested your commit with a worse quality camera and I'm not sure there's any difference. It's kind of hard to tell but I tested by starting a stop watch timer app on a smartphone and pointing it at the camera while recording. At first it seemed like the worse quality camera stream was a bit slower than before, but if so then by a seemingly negligible amount.

NinjaWeeaboo, 2020-01-04:
Shoot, thanks for testing anyways.

vashmata, 2020-01-11:
>Also this makes me wonder if we should maybe look into getting and displaying current FPS of the video stream

Yes pls it's gamer time

closed 2022-10-31 by VimFreeman