space-concordia-robotics/robotics-prototype #247: Branch name verification script blocks ability to use git rebase

PeterGhimself, 2019-12-20:
I was trying to edit the commit message of `HEAD^` and when I tried I got the following output:
```(venv) beep@boop:~/Programming/robotics-prototype$ git rebase -i HEAD~2
[detached HEAD 1bf0811] [#151][#244] Task handler portion allows simultaneous streams
 Date: Wed Dec 18 16:43:58 2019 -0500
 1 file changed, 57 insertions(+), 27 deletions(-)
fatal: ref HEAD is not a symbolic ref
Branch name is not named properly, please see wiki for formating: https://github.com/space-concordia-robotics/robotics-prototype/wiki/Work-Flow
Could not apply cbcd8f246dd1017a95d90b58682151e38c13b5eb... [#151][#244] Update Listener class, start_ros_stream script and task handler to allow simultaneous camera streams
(venv) beep@beep-Inspiron-15-7000-Gaming:~/Programming/robotics-prototype$```

When rebasing you end up checking out a new branch which gets named in the format `(no branch, rebasing simultaneous-streams-151)` for example, so this branch name pattern should allowed.

PeterGhimself, 2019-12-20:
For now the workaround is to temporarily disable the hook

MartensCedric, 2021-01-30:
also fails to work with vim-fugitive

MartensCedric, 2021-01-30:
same reason I believe

closed 2022-10-31 by VimFreeman