space-concordia-robotics/robotics-prototype #249: Stream offline placeholder image does not initially appear

PeterGhimself, 2019-12-25:
On start up, since the video streams are offline by default we usually display a "stream-offline" image placeholder. It seems that the image does not appear but instead the space is just taken up by the background color. After succesfully turning ON and then OFF a stream the place holder image reappears. Refreshing the page though will cause it to re-disappear if the stream is OFF (after having been turned ON once, so the placeholder image was there). No js console errors appear.

NinjaWeeaboo, 2019-12-29:
This seems to be fixed now in new-camera-template-201

PeterGhimself, 2019-12-29:
@NinjaWeeaboo my bad for not being specific enough how to reproduce the bug properly.
This is caused when the `web_video_server` is running but no `cv_camera` nodes are running.
So if you are testing locally it will not be reproduced unless you are running `web_video_server`.
After PR #248 will be merged testing locally will be made easier by running `roslaunch local_rover.launch`

NinjaWeeaboo, 2019-12-30:
Ah I see, this issue does still occur in new-camera-template-201. My bad.

closed 2020-02-16 by MartensCedric