space-concordia-robotics/robotics-prototype #251: Activate rover toggle button does not initialize properly on startup

PeterGhimself, 2019-12-25:
To reproduce we need to either have the GUI running with the actualy rover connected or just a teensy running the code for the mobile platform/wheels micrcontroller.

Steps to reproduce:
1. Activate the command listener for the rover
2. Send "activate" command to MCU
3. Confirm by either moving the rover wheels with WASD keys or look for responses from serial monitor in Arduino IDE if testing locally (if there are none they need to be added for increased transparency/easier debugging).
4. Refresh the rover page and the command listener will toggle ON but the activate control button will stay OFF.

This is annoying because now to deactivate it throught the GUI we need to hack the button to be in "ON" mode visually, which can be done manually by running something like `$('#toggle-button-id-of-that-button').checked = true` in the js browser console, but it is only an annoying workaround which does not solve the issue.

vashmata, 2020-01-11:
Perhaps like with #237 we can use ROS parameters for active status. And maybe even for listener status?

MartensCedric, 2020-02-14:
From a further investigation, it appears that the rovernode crashes and needs to be manually restarted.

MartensCedric, 2020-02-25:
Sometimes when you launch a node from the shell you get this crash : 

INFO] [1582662526.320018]: Using 115200 baud by default
[INFO] [1582662526.320712]: Using USB and 115200 baud by default
[INFO] [1582662526.380984]: 2 USB device(s) detected
[INFO] [1582662526.381903]: Attempting to connect to /dev/ttyACM1
[INFO] [1582662526.385298]: clearing buffer...
[INFO] [1582662526.387379]: identifying MCU by sending 'who' every 300 ms
[INFO] [1582662526.388066]: attempt #1...
[INFO] [1582662526.389724]: response: ASTRO GOT: ~

[INFO] [1582662526.392947]: response: ASTRO BOOOOOOO!!! Wrong command, try a different one, BIAAAAAA

[INFO] [1582662526.445465]: response: ASTRO Battery voltage: 15.31

[INFO] [1582662526.482476]: response: ASTRO HEADING-OK 255.30 -- GPS-N/A

Traceback (most recent call last):
  File "/home/odroid/Programming/robotics-prototype/robot/rospackages/src/mcu_control/scripts/ArmNode.py", line 200, in <module>
    init_serial()
  File "/home/odroid/Programming/robotics-prototype/robot/rospackages/src/mcu_control/scripts/ArmNode.py", line 73, in init_serial
    response = ser.readline().decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf0 in position 12: invalid continuation byte

vashmata, 2020-02-25:
Can we remove the critical label?

It boils down to when an MCU node attempts to establish a serial connection with the relevant MCU. You'll find similar problems with the other MCU node scripts. I believe it's got something to do with the fact that we have multiple usb devices connected to the odroid at the same time.

It's possible that it's more specifically due to the fact that they're connected via a usb hub. Keep in mind that there's the external usb hub we're using but also an internal one in the odroid's circuitry.

Regardless of what the issue is, if there's a way to make this robust it could be nice, but it's not critical unless we become unable to run the MCU node no matter how many times we try.

MartensCedric, 2020-02-25:
Just pasting a different crash : Maybe related, maybe not : INFO] [1582665360.080515]: got raw data: ASTRO Linear Velocity: -0.00 m/s RotaASTRO GOT: 26:0

[WARN] [1582665360.138258]: GPS data unavailable
[INFO] [1582665360.143210]: received: 27:0 command from GUI, sending to rover Teensy
[INFO] [1582665360.156355]: got raw data:  Desired Velocities: 135.31, 135.31, 135.31, 135.31, 135.31, 135.31

[INFO] [1582665360.215620]: received: 28:0 command from GUI, sending to rover Teensy
[INFO] [1582665360.286379]: received: 29:0 command from GUI, sending to rover Teensy
[WARN] [1582665360.344597]: GPS data unavailable
[INFO] [1582665360.353643]: received: 30:0 command from GUI, sending to rover Teensy
[INFO] [1582665360.360088]: got raw data:  Desired Velocities: 150.92, 150.92, 150.92, 150.92, 150.92, 150.92

[INFO] [1582665360.422127]: received: 31:0 command from GUI, sending to rover Teensy
[INFO] [1582665360.429715]: got raw data: ASTRO LinASTRO GOT: 31:0

[INFO] [1582665360.493040]: received: 32:0 command from GUI, sending to rover Teensy
[INFO] [1582665360.562823]: received: 33:0 command from GUI, sending to rover Teensy
Traceback (most recent call last):
  File "/home/odroid/Programming/robotics-prototype/robot/rospackages/src/mcu_control/scripts/RoverNode.py", line 322, in <module>
    publish_nav_states(feedback)
  File "/home/odroid/Programming/robotics-prototype/robot/rospackages/src/mcu_control/scripts/RoverNode.py", line 204, in publish_nav_states
    heading,gps = message.split(' -- ')
ValueError: too many values to unpack (expected 2)
(venv) odroid@odroid:~$

vashmata, 2020-02-25:
yeah after this is over we need to go and fix my bad python job lol

MartensCedric, 2020-02-25:
@vashmata 

>Can we remove the critical label?

This problem makes the GIT fail, thus it is critical to fix
The expected behavior for the operator is to activate the rover via the GUI not to SSH into the rover.

MartensCedric, 2020-02-25:
Although the other python issues are probably unrelated so Ill make a new issue

closed 2022-10-31 by VimFreeman