space-concordia-robotics/robotics-prototype #253: Inconsistent feedback on PDS individual motor cut/provide power toggle buttons

PeterGhimself, 2019-12-25:
The buttons do succesfully send the proper commands and the PDS will succesfully receive them and act accordingly. However on the GUI side of things the buttons do not consistently change their states in an appropriate manner.

For instance during my testing today of turning OFF each motor one by one from 1 to 6, the first two buttons did not change their state to "OFF" even though the motors would not move when sending forward drive commands. The rest of individual motor power toggle buttons did indeed switch to "OFF" properly. But then when going through all those "OFF" buttons to send the "ON" signal, even though the PDS responded correctly those buttons remained "OFF".

I think this issue may have something to do with the way that the responses of the PDS are being parsed, since during testing I noticed in the console log that sometimes there seemed to be two separate messages that ended up being merged into one. For example in the console log we would see something like (this is not exact output):

```
motor 1 - 0
motor 2 - 0
motor 3 - 0
<some message that doesn't belong here, or should at least not interrupt the PDS response>
motor 4 - 0
motor 5 - 1
motor 6 - 1
```

PeterGhimself, 2019-12-25:
Additionally, when solving this issue we might as well incorporate these checks on the initialization of the page so that we can instantly see after refreshing the correct statuses of all the motors.

vashmata, 2020-01-11:
>I think this issue may have something to do with the way that the responses of the PDS are being parsed

I agree, I think it's something related to UART or the parsing or maybe how fast the messages are sent or something.

closed 2022-10-31 by VimFreeman