space-concordia-robotics/robotics-prototype #256: GPS straight line goal planner action

PeterGhimself, 2019-12-29:
## Problem
To start with our first autonomy module we need a simple straight line goal planner with the ultimate purpose of driving the rover from it's current GPS coordinate to a destination coordinate providate by the client.

## Sub tasks
- [ ] Setup a boilerplate ROS action with dummy data for POC (don't do actual navigation logic yet, just have like `print("feedback 1"); sleep(1);print("feedback 2");sleep(1); etc.`, like a template
- [ ] Implement skeleton code based of design for decision making process during path planning
- [ ] Add drive commands from #209 
- [ ] Functional testing

vashmata, 2020-02-07:
This past week we succeeded in running the actionlib beginner tutorial. Continuing forwards the autonomy team will continue learning about actions and attempt to create one themselves.

KoaTheKoala, 2020-03-08:
I added two ROS Action/Server demos for reference. The first one is just based off of the ROS action tutorial linked here: http://wiki.ros.org/actionlib_tutorials/Tutorials

The first demo is for calculating a set of numbers in the Fibonacci sequence.
The second demo is for 'washing dishes' by controlling a for loop control variable.

See the readme in robotics-prototype/robot/demos/RosAction/ for more information. The long version of the readme will contain condensed steps to building an action from scratch. It's partially done. I will finish it soon. Otherwise, we are ready to go to start building custom actions -insert sunglasses emoji-

vashmata, 2020-03-08:
@PeterGhimself @MartensCedric I find it a bit weird that we would keep ROS demos in a place where we can't actually run them. I would propose we make a package called Demos within rospackages where we hold them so that people can run themm right away

vashmata, 2020-03-08:
I have a few other demo ROS scripts that I need to relocate as well

closed 2020-06-09 by MartensCedric