space-concordia-robotics/robotics-prototype #262: Test ar_track_alvar ROS package

KoaTheKoala, 2019-12-31:
- [ ] Install ROS Kinetic
- [ ] Install ar_track_alvar ROS package
- [ ] Print 5-10 AR tags (you may need to use the ar_track_alvar package or something similar to
- [ ] Test ar_track_alvar ROS package
- [ ] Test distances (these distances can be approximate, we just need a rough idea)
- - [ ] Test 1 m
- - [ ] Test 3 m
- - [ ] Test 5 m
- - [ ] Test 10 m
- - [ ] Test 15 m
- [ ] Test angles (these angles can be approximate, we just need a rough idea)
- - [ ] Test 0 degrees (straight on)
- - [ ] Test 15 degrees
- - [ ] Test 30 degrees
- - [ ] Test 45 degrees
- - [ ] Test 60 degrees
- - [ ] Test 75 degrees
- [ ] Test lighting

vashmata, 2020-02-07:
URC reminders:
1) Every single post at URC will have its own AR tag, so we'll be able to know which exact post we're looking at based on the AR tag. See 5.Q in the [URC Q&A](http://urc.marssociety.org/home/q-a)
2) According to the URC rules, the posts will have AR tags in the form of a triangular prism constructed of 3 identical AR tags. This means that it's likely for the camera to see the AR tag on two faces of the prism simultaneously. We should test this scenario as well.

MartensCedric, 2020-07-11:
is this investigated? @KoaTheKoala

MartensCedric, 2020-09-12:
Can we close this @KoaTheKoala @domludera  I'm gonna open a new issue to transfer ERC code

KoaTheKoala, 2020-09-20:
@MartensCedric I think it is safe to close this issue, but imo, we should still at least test under different lighting and distance conditions.

closed 2020-11-28 by MartensCedric