space-concordia-robotics/robotics-prototype #264: Camera Stream Settings

TheRealJeff, 2020-01-01:
Stream settings should be investigated to reduce the bandwidth required per stream. Parameters that should be looked in modifying are:
* Color Space: RGB vs. YUV 400, 420, 422, 444
* Color Depth: 8-bit, 10-bit, 12-bit
* Resolution

Playing around with these paramters can permit to find a selection of them to reduce the bandwidth requires such that we can have the maximum resolution.

Another thing to look into is adaptive resolution, which is to say to that perhaps we can have these stream parameters:
* 1 stream -- 1080p
* 2 streams -- 720p
* 3 streams -- 240p
These are just examples however

MartensCedric, 2020-01-03:
We also need to know what is the best combination of stream settings. In order words, suppose we know that we can run 3 streams at 480 but only 1 at 720. Can we run 2 480 with 1 at 720?

vashmata, 2020-01-11:
This reminds me of what @Davidster brought up at PDR. Perhaps we can at some point come up with a system that will track the bandwidth used by the cameras and automatically adjust individual camera settings in order to make sure they don't take up all of the bandwidth and cause lag and issues with controlling the rover.

closed 2022-10-31 by VimFreeman