space-concordia-robotics/robotics-prototype #265: PDS multi motor commands not working

PeterGhimself, 2020-01-02:
From testing with the GUI it seems that the PDS is responsive to the commands controlling indiviudal motors (cut or provide power), but not to the commands addressing all motors at once.

The console log output is garbled, meaning that in between the PDS feedback we see other responses or messages inserted. It is currently unclear whether this is just a pure frontend issue or perhaps an issue on the lower level.

PeterGhimself, 2020-01-02:
[This is the dirty script](https://github.com/space-concordia-robotics/robotics-prototype/commit/064d744cad907c42994b5bf4c874f91943e96e72) I've been using for testing uart comms in the past, and it works for usb or uart the same you just provide it the appropriate port value, baudrate and message and it will keep sending that value over the port and listen for and print responses.

vashmata, 2020-01-11:
Also related to #252 I'm guessing. My intuition says that it's some weird thing with the js.

MartensCedric, 2020-02-08:
#252 is fixed with #286 . We should test for #265 once the wheels are back on the rover

closed 2022-10-31 by VimFreeman