space-concordia-robotics/robotics-prototype #270: Readd motor class unit test

PeterGhimself, 2020-01-06:
In [this](https://github.com/space-concordia-robotics/robotics-prototype/commit/2d7c76e7f89b7f6a14a655c3fd4fe24c92465979) commit we removed the motor_test unit test since we removed a class `motor.py` which was in `basestation` folder since we though it wasn't being used.

While it may have no functional use for the project, it is a good proof that our repo built correctly in that `setup.py develop` worked properly and we can import our own custom modules with the `robot.basestation.etc` format.

The test should be readded as well as the class, perhaps it should be placed in a different folder though.

- [ ] Locate missing motor.py source code
- [ ] Decide appropriate place to put it back
- [ ] Readd unit test

MartensCedric, 2020-01-07:
Since the purpose of this issue is not _actually_ to add `motor.py` back but to actually see if the imports are correct. I suggest that instead of adding a seemingly useless class back as a unit test we create a new unit test that validates that the imports are done correctly. This is also for the future developers on the project, which will immediately see its importance instead of doubting its utility.
@PeterGhimself

PeterGhimself, 2020-01-07:
Fair enough @MartensCedric

PR #271 (2020-01-09): Custom module import statement test
# Assignee Section

## Description
A simple test for verifying that `python setup.py develop` or `pip install -e` work correctly during project build phase and allow us to import our own custom modules.

closes #270

Approval from at least one software team lead is necessary before merging.

# Reviewer Section

Aside from local testing and the General Integration Test it is implied that static analysis should be included in the verification process.

- [x] Local Test Performed Successfully
- [ ] [General Integration Test](https://docs.google.com/document/d/1 [... cut]
closed 2020-01-09 by PeterGhimself